OnionIoT/Onion-Hardware #3: Onion Omega2+ and Power Dock Schematics

jorgegarciadev, 2017-01-11:
Any plans on releasing the schematics for the Onion Omega2+ and the Power Dock? And if it's going to take a long time... can I power the Power Dock vía +5v connector on it?

kzyapkov, 2017-01-20:
+1 on the power dock schematics. I'd like to hack it with a low-power MCU to intermittently wake up the Omega and run the setup for months off a fat battery.

Market1G, 2017-05-22:
It's obviously made to be powered from the +5V USB connector but what are the power requirements and usage? What can be the RF interference with WiFi? How low can go the power supply +4.75, +4.5V, lower?

AlexDaniel, 2017-08-04:
Any news? Should we even hold our breath for this?

Testato, 2017-08-16:
It is absolutely unjustified that after six months the files are not yet publicized.
Is this what you mean for opensource?
Here you must immediately release at least the .pdf of the wiring diagram otherwise no one will buy the card any more.

greenbreakfast, 2017-10-13:
Hi, the Power Dock schematic has been uploaded. And yes, it can be powered with the 5V and GND pins on the Expansion Header.

We will not be releasing the schematics of the Omega2 and Omega2+. We have never promised to do so, it was only the original Omega1 that we open sourced.

AlexDaniel, 2017-10-13:
@ greenbreakfast OK, thank you for your response.

5N44P, 2017-10-13:
> We will not be releasing the schematics of the Omega2 and Omega2+. We have never promised to do so, it was only the original Omega1 that we open sourced.

So you should really update the [About section](https://onion.io/about/) of your website. 
It states: 
> We are strong believers of open source and we strive to give back to the open source community. **All hardware schematics, board design, and bill of materials will be made available to the community.**

Testato, 2017-10-13:
you have proven to be the classic company that only wants to exploit the world tied to the concept of opensource and then betray it at the first opportunity. There is nothing to hide in a project like the Onion, you just lost your face.

kzyapkov, 2017-10-14:
@greenbreakfast what's the deal with the missing part refs?

pchidley, 2018-10-02:
Where are the part numbers on the Power Dock 2 schematic?

Specifically the battery manager and buck converter?